openrisc/mor1kx #29: Edge triggered IRQ does not work as expected

wwwweb, 2015-04-27:
When using edge triggered IRQs, the (rising) edge seems to be detected using the PICSR itself and the incoming IRQLINE  (!spr_picsr[irqline] & irq_unmasked[irqline] in mor1kx-pic.v). When resetting the according IRQ bit in PICSR with the IRQLINE still high (no new edge but still high level), the IRQ is triggered in the next cycle again, so this just works like a level triggered IRQ.

From my opinion there needs to be an additional register stage for edge detection which only generates a 1 cycle set. Using this, I think the logic that "CLEAR" is served before "SET" is not sufficient, as I would expect that when an edge occurs in the instant I clear the IRQ (usually within the ISR), the IRQ bit stays active and another IRQ is triggered.

Best regards,
wwwweb

skristiansson, 2015-04-27:
I agree, it doesn't look right. I'll look into it closer.

skristiansson, 2015-05-14:
This should be fixed by:
https://github.com/openrisc/mor1kx/commit/6caa7dba90cbe7c51421b542bd7b5ed487c83195

Please test and confirm.

wwwweb, 2015-06-08:
Hi skristiansson,

I had a look at the bugfix. I think it is not fully fixed yet. Assuming, that the according irqline is high when enabling the irq in picmr, the check on the rising edge is true immediately, although this rising edge may have occurred when the irqline was masked or may not have occurred at all if irqline is pulled up by default. I think, irq_unmasked_r <= irq_i may work better in the registered process.

Best regards,
wwwweb

skristiansson, 2015-06-10:
yes, agreed, the registering need to take place before the unmasking.