openrisc/mor1kx #30: Problem with 2 IRQs at the same time

wwwweb, 2015-04-30:
I am not sure if this is a problem with mor1kx or the or1k-support library.

When enabling 2 IRQs (i.e. IRQ1 and IRQ2), registering an ISR for each (i.e. ISR1 and ISR2) and then triggering both IRQs at the same time (i.e. PICSR becomes 0x00000006), the program "hangs up" after serving ISR1.

When using LEVEL triggered IRQs, ISR1 is repeated until the trigger condition is negative, then the program hangs.

When using LATCHED_LEVEL triggered IRQs and asserting the trigger only for one clock cycle, ISR1 is executed one time and then the program hangs. With the same setting, when doing or1k_interrupts_disable() at the beginning of ISR1 and or1k_interrupts_enable() at its end, the program enters ISR2 after serving ISR1 and hangs up.

I am using mor1kx_cappuccino. When triggering the IRQs one after the other (so there is only one 1 IRQ active at a time), both ISRs work and return correctly.

"hangs up" does not mean that it really hangs but at least it does not return to the correct point in the main program. It also seems that mor1kx_ctrl_cappuccino is trying to recover from some state repeatedly.

Best regards,
wwwweb

wallento, 2015-04-30:
Hi,

I think this is a bug in the libgloss port. Can you please verify that it still exists with the current development version at openrisc/newlib? There has been a patch some weeks ago.

Best,
Stefan

wwwweb, 2015-05-04:
Hi Stefan, all,

I tried with the current dev version of newlib/libgloss but it still does not return from the ISR.

I could provide more information if needed but I am not sure which makes sense.

Best regards,
wwwweb

wallento, 2015-05-06:
Hey,

do you by chance have a short list to reproduce your scenario at hand? Otherwise I unfortunately need a few more days to fix this.

Best,
Stefan

wwwweb, 2015-05-06:
Hi Stefan,

Thanks for your reply. I did some further investigation and it seems, that the problem may not be with the IRQ mechanism itself but with printf() in combination with IRQs.

What I have done is the following (pseudocode):
```
int main()  {
  
  int i = 0;  
  
  printf("Here we go... \n")
  
  //enable IRQs
  or1k_interrupt_handler_add(1, or1k_isr_1, (void *)0);
  or1k_interrupt_handler_add(2, or1k_isr_2, (void *)0);
  or1k_interrupt_enable(1);
  or1k_interrupt_enable(2);
  or1k_interrupts_enable();
  
  //main loop
  while(1) {
    printf("loop number %u \n", (unsigned int)i);
    i++;
  }
  
  return 0;
}

void or1k_isr_1(void* data) {  
  printf("IRQ1\n");
  or1k_mtspr(OR1K_SPR_PIC_PICSR_ADDR, 0x00000002);
  return;
}

void or1k_isr_2(void* data) {  
  printf("IRQ2\n");
  or1k_mtspr(OR1K_SPR_PIC_PICSR_ADDR, 0x00000004);
  return;
}
```

When triggering both (external) IRQs after some time at the same time, the output (Modelsim simulation with the UART16550 in simulation mode) looks like this (in case the IRQs are triggered only once):

Here we go...
loop 0
loop 1
loop 2
IRQ1
IRQ2
[nothing more]

or sometimes also

Here we go...
loop 0
loop 1
loo
IRQ1
IRQ2
p 2
[nothing more]

dependent on the time the IRQs are triggered. In Modelsim can be seen, that Wishbone Cyc to the UART never becomes active any more (but the bus is still active, not blocked), so I assume it is no failure of the UART model. Moreover, when I change the main to i.e.
```
int main() {
  
  uint32_t volatile * wb_addr = 0x91000000;
  uint32_t a = 0x00000000;
  
  ...[same as before]
  
  
  while(1) {
    printf("loop number %u \n", (unsigned int)i);
    i++;
    
    *(wb_addr) = a;
    a++;
  }
  
  return 0;
}
```
the UART output remains the same as above but the wb_address 0x91000000 is still incremented when the ISRs have returned. From tracelog, it can be seen, that printf is also called continuously but its trace is a lot shorter than before the ISRs where it really printed sth. , so I assume it is in a dysfunctional state (i.e. __sprint_r is never called in the late printf's compared to the functional printf's).

This behavior is monitored with the current dev version of newlib/libgloss. With the older version I had before, after the ISRs the code runs into a trap exception which is not the case with the new library but obviously there is still sth. wrong.

Could you make an educated guess if this is only a problem of printf or if it also may occur with other IRQs + other functions? The first case would be okay for now, the second seems more problematic.

Best regards,
wwwweb

wallento, 2015-05-06:
Hey,

the second output from your loop example makes me suspicious that it may be an issue with the restored link register. As the characters are printed, the exception handler returns.
I will pm you for debugging.

Best,
Stefan

wallento, 2015-05-08:
The problem is fixed by https://github.com/openrisc/newlib/commit/fb3f79691e2268700a52870ac7a4a227d5a56806

closed 2015-05-08 by wallento