openrisc/mor1kx #91: mor1kx_monitor terminate Multisim simulation

lorenzdbs, 2019-08-03:
Hi, 

I'm a newbie in Verilog coding, so most probably this is the cause of my problem, but, working on a university project that include the mor1kx module [(https://github.com/mit-ll/CEP)](https://github.com/mit-ll/CEP), I'm not able to run a complete simulation of the architecture because it stops at line 144 of the mor1k_monitor file.
So I'd like to ask what is that _execute_insn_ variable that bring my simulation to finish and, if it is possible to know, why it has that value.
Thank you in advance to all those who can help me,

Lorenzo

stffrdhrn, 2019-08-03:
Hi, 

This is expected behavior.  The operation `l.nop 0xC` is a special instruction to cause the simulator to exit.  We do this for verilog and software simulators.  Why are you running this instruction and not expecting it to exit?

## Details

On line [144](https://github.com/openrisc/mor1kx/blob/master/bench/verilog/mor1kx_monitor.v#L140) I see this:

```
	  if (execute_insn == 32'h15_00_00_0c)
	    begin
	       // Silent exit
	       $finish;
	       
	    end
```

If you look at our [architecture spec](https://openrisc.io/architecture) I see:

Opcode `0x15` is an `l.nop K`.  With description

*This instruction does not do anything except that it takes at least one clock cycle to complete. It is often used to fill delay slot gaps. Immediate value can be used for simulation purposes.*

As it mentions the K option can be used for simulation purposes.  As we see here the argument `0xc` causes the simulator to exit.

lorenzdbs, 2019-08-04:
I was wondering what was making the execute_insn variable to be equal to 32'h15_00_00_0c.

But, like you said, if I don't want to end the simulation I just don't need this instruction: I though that the other "if" conditions were executing some important code, but they only include display functions so I can just comment it all. I let you know if the simulation will proceed correctly.
Thanks for the useful informations! ;)

stffrdhrn, 2019-08-04:
BTW, what is the program you are running? Do you have a link?  Why do you think exiting is wrong?  It's probably good to understand what the program is trying to do.

lorenzdbs, 2019-08-04:
I'm running with Multisim the run_sim_CEP.do file (in the repository I linked in the previous comment, "simulation" directory) through the terminal command line

`vsim -do run_sim_CEP.do`

The aim is to see if I can simulate specific cores disabling them in the _orpsoc-defines.sv_ file without running the single _.do_ files of each core (I've already launched them and they passed the tests).
Now that I've commented the part of the program in _mor1kx_monitor.v_ the simulation has another problem: it doesn't stop. In the _cep_regression.cpp_ file you can see that at line 1136 an AES core test starts: before writing this issue the simulation was stopping right after this part started, now it starts but never finish.
I attach what is written in the Multisim transcript. 

[Modelsim simulation.pdf](https://github.com/openrisc/mor1kx/files/3464923/Modelsim.simulation.pdf)

Thanks for the help.

lorenzdbs, 2019-08-04:
@stffrdhrn if you had the need to install the repository to test it I can send you a doc file with a guide to create it it without problems

stffrdhrn, 2019-08-04:
> I'm running with Multisim the run_sim_CEP.do file (in the repository I linked in the previous comment, "simulation" directory) through the terminal command line
>
> `vsim -do run_sim_CEP.do`
>
> The aim [...]

You probably should not be commenting out the `mor1kx_monitor.v` code.  The line
you commented allows software to exit, it should not be commented out.

I am guessing that in the AES routine you are getting an error and the system
exits.  You might want to start up the debugger and trace what the issue is.
ANother thing you can do is try to add some printf() statements to see where in
the AES code the system is issuing the exit() function and why.

-Stafford

lorenzdbs, 2019-08-04:
Ok, thank you so much for the help I'll try what you suggested.
One last thing, what you mean by "start up the debugger"? I found something [here](http://www.tkt.cs.tut.fi/tools/public/tutorials/mentor/modelsim/getting_started/gsms.html#debugging) where is explained how to debug in multisim by adding breakpoints etc. Is this what you meant? (sorry for such probably dummy questions but I'm trying to get the hand of it).

stffrdhrn, 2019-08-04:
Sorry I don't think so. I mean debugging the software with gdb.  I haven't
done that with modelsim before.  It's probably easier for you to use
printf"s.

lorenzdbs, 2019-08-05:
Got it, thank you!

stffrdhrn, 2019-12-03:
Will choose

closed 2019-12-03 by stffrdhrn