openrisc/mor1kx #95: bug in LWA, SWA and snooping hit

hurjaewon, 2019-10-29:
After LWA instruction, if a snoop_en_i signal and snoop to same address are inserted at some specific time, the reservation does not get cancelled.
According to OpenRISC isa, they said snooping hit should break the reservation. 
Isn't it bug?

Here's waveform
[image]

stffrdhrn, 2019-11-06:
Hi @hurjaewon ,

Thanks for the report, it sounds like a bug.  Do you have any code/setup to reproduce it?  If so it would be helpful.

stffrdhrn, 2019-11-06:
In your trace can you provide signals for:

 - mor1kx_lsu_cappuccino . dbus_adr_o
 - mor1kx_lsu_cappuccino . dbus_ack_i
 - mor1kx_lsu_cappuccino . snoop_valid

Do you know where the second snoop to 0x2000 is coming from? Are you sure it is external?

JaewonHur, 2019-12-02:
Sorry, I missed to reply. 
Actually I'm trying with one mor1kx cpu and manually making inputs. 
So there is only one snoop as in the figure and other signals are like this.

I was testing with DATA_CACHE=ENABLED, and all the codes are as same as the master branch.
Thank you.

[image]