openrisc/mor1kx #96: [Bug] jalr to link register

JaewonHur, 2019-12-02:
When mor1kx receives 'jalr to link register (r9)' 
illegal instruction exception should be asserted as in or1ksim, but it doesn't

in mor1kx_decode.v, changing as below should fix this.

always @*
 case (opc_insn)
  `OR1K_OPCODE_JALR:
   if (decode_insn_i[`OR1K_RB_SELECT] == 9)
    decode_except_illegal_o = 1'b1;

stffrdhrn, 2019-12-03:
This makes sense. But it doesn't seem to be in the spec to assert the exception in this case.  The spec says:

>  It is not allowed to specify link register r9 (see Register Usage on page 335) as rB.
This is because an exception in the delay slot (including external interrupts) may cause
l.jalr to be reexecuted.

It's not clear that the exception assertion is correct.  Do you have any thoughts?

JaewonHur, 2019-12-03:
According to the spec, prohibited instruction may leave the cpu state in undefined state.

In this case, I think whether asserting exception or not depends on the designer.

stffrdhrn, 2019-12-03:
Yes. In that case we can add this.