openrisc/mor1kx #101: [Bug] Adder carryout bug

JaewonHur, 2019-12-09:
@lifeasageek

mor1kx adder mistakenly set 'adder_carryout' when subtract.
You can reproduce the case with below instructions.

```
18000000 l.movhi r0,0x0000       r0         = 00000000  flag: x
a8200001 l.ori   r1,r0,0x0001    r1         = 00000001  flag: x
c0000811 l.mtspr r0,r1,0x0011    SPR[0011]  = 00000001  flag: x
a8800004 l.ori   r4,r4,0x0004    r4         = 00000000  flag: 0
a8600003 l.ori   r3,r0,0x0002    r3         = 00000000  flag: 0
e0441802 l.sub   r2,r4,r3        r2         = 00000000  flag: 0
```
My solution is,
in mor1kx_execute_alu.v
change
```verilog
  wire [OPTION_OPERAND_WIDTH-1:0] b_neg;
  wire [OPTION_OPERAND_WIDTH-1:0] b_mux;

  assign b_neg = ~b;
  assign b_mux = adder_do_sub_i ? b_neg : b;
```

to 
```verilog
  wire [OPTION_OPERAND_WIDTH:0] b_neg;
  wire [OPTION_OPERAND_WIDTH:0] b_mux;

  assign b_neg = ~{{1{1'b0}}, b};
  assign b_mux = adder_do_sub_i ? b_neg : {{1{1'b0}}, b};
```

skristiansson, 2019-12-19:
I don't understand the bug description, the cy flag should be set when an unsigned overflow occurs.
Are you saying that that is not happening?

JaewonHur, 2019-12-23:
Sorry, I misunderstood unsigned overflow.
Associated pull requests can be ignored.
Closing.

closed 2019-12-23 by JaewonHur