openrisc/mor1kx #104: [Bug] mtspr shadow gpr read

JaewonHur, 2019-12-13:
@lifeasageek

When reading or writing shadow gprs through mfspr (mtspr),
it seems mor1kx_rf_cappuccino doesn't offset the register address.
(As the spec, SPR[1024+x] points to GPR_'x')

You can reproduce error through below instructions
```
18000000 l.movhi r0,0            
a8200001 l.ori   r1,r0,0x1       
c0000811 l.mtspr r0,r1,0x11      
15000000 l.nop   0               
c0010c0e l.mtspr r1,r1,0x40e 
```

Changing the code in mor1kx_rf_cappuccino.v as follow should solve this issue.
```verilog
  assign rf_wradr = wb_rf_wb_i ? wb_rfd_adr_expand : (spr_bus_addr_i[RF_ADDR_WIDTH-1:0] - `RF_ADDR_WIDTH-1'h400);
```

stffrdhrn, 2019-12-14:
Thanks @JaewonHur for these big reports and fixes.  You seem to have fixes for many of them. Do you want to open a pull request for the fixes so they can run in our test suite?

stffrdhrn, 2019-12-14:
Sorry I haven't had time to get through to checking all of them.

skristiansson, 2019-12-19:
The offset is already handled at https://github.com/openrisc/mor1kx/blob/master/rtl/verilog/mor1kx_rf_cappuccino.v#L262

closed 2019-12-19 by skristiansson