openrisc/mor1kx #125: Librecores CI failing

stffrdhrn, 2021-06-25:
The librecores CI has been failing for some time, we now have setup github actions.  Should we remove librecores CI?

stffrdhrn, 2021-06-25:
From the logs it fails cloning with operation not permitted, it seems like there is a filesystem issue.

https://ci.librecores.org/job/Projects/job/OpenRISC/job/mor1kx/job/master/19/console

```
Cloning repository https://github.com/openrisc/mor1kx.git
ERROR: Failed to clean the workspace
jenkins.util.io.CompositeIOException: Unable to delete '/var/lib/jenkins/workspace/Projects_OpenRISC_mor1kx_master'. Tried 3 times (of a maximum of 3) waiting 0.1 sec between attempts. (Discarded 54 additional exceptions)
	at jenkins.util.io.PathRemover.forceRemoveDirectoryContents(PathRemover.java:90)
	at hudson.Util.deleteContentsRecursive(Util.java:261)
	at hudson.Util.deleteContentsRecursive(Util.java:250)
	at org.jenkinsci.plugins.gitclient.CliGitAPIImpl$2.execute(CliGitAPIImpl.java:743)
	at hudson.plugins.git.GitSCM.retrieveChanges(GitSCM.java:1132)
	at hudson.plugins.git.GitSCM.checkout(GitSCM.java:1177)
	at org.jenkinsci.plugins.workflow.steps.scm.SCMStep.checkout(SCMStep.java:125)
	at org.jenkinsci.plugins.workflow.steps.scm.SCMStep$StepExecutionImpl.run(SCMStep.java:93)
	at org.jenkinsci.plugins.workflow.steps.scm.SCMStep$StepExecutionImpl.run(SCMStep.java:80)
	at org.jenkinsci.plugins.workflow.steps.SynchronousNonBlockingStepExecution.lambda$start$0(SynchronousNonBlockingStepExecution.java:47)
	at java.util.concurrent.Executors$RunnableAdapter.call(Executors.java:511)
	at java.util.concurrent.FutureTask.run(FutureTask.java:266)
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1149)
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:624)
	at java.lang.Thread.run(Thread.java:748)
	Suppressed: java.nio.file.FileSystemException: /var/lib/jenkins/workspace/Projects_OpenRISC_mor1kx_master/build/mor1kx_5.0-r3/synth-icestorm/mor1kx_5.0-r3.eda.yml: Operation not permitted
		at sun.nio.fs.UnixException.translateToIOException(UnixException.java:91)
		at sun.nio.fs.UnixException.rethrowAsIOException(UnixException.java:102)
		at sun.nio.fs.UnixException.rethrowAsIOException(UnixException.java:107)
		at sun.nio.fs.UnixFileAttributeViews$Posix.setMode(UnixFileAttributeViews.java:238)
		at sun.nio.fs.UnixFileAttributeViews$Posix.setPermissions(UnixFileAttributeViews.java:260)
		at java.nio.file.Files.setPosixFilePermissions(Files.java:2045)
		at jenkins.util.io.PathRemover.makeWritable(PathRemover.java:286)
		at jenkins.util.io.PathRemover.makeRemovable(PathRemover.java:259)
		at jenkins.util.io.PathRemover.removeOrMakeRemovableThenRemove(PathRemover.java:239)
		at jenkins.util.io.PathRemover.tryRemoveFile(PathRemover.java:205)
		at jenkins.util.io.PathRemover.tryRemoveRecursive(PathRemover.java:216)
		at jenkins.util.io.PathRemover.tryRemoveDirectoryContents(PathRemover.java:226)
		at jenkins.util.io.PathRemover.tryRemoveRecursive(PathRemover.java:215)
		at jenkins.util.io.PathRemover.tryRemoveDirectoryContents(PathRemover.java:226)
		at jenkins.util.io.PathRemover.tryRemoveRecursive(PathRemover.java:215)
		at jenkins.util.io.PathRemover.tryRemoveDirectoryContents(PathRemover.java:226)
		at jenkins.util.io.PathRemover.tryRemoveRecursive(PathRemover.java:215)
		at jenkins.util.io.PathRemover.tryRemoveDirectoryContents(PathRemover.java:226)
		at jenkins.util.io.PathRemover.forceRemoveDirectoryContents(PathRemover.java:87)
		... 14 more
```

closed 2021-06-25 by stffrdhrn

oleg-nenashev, 2021-06-25:
Likely. We do not have contributors who have time to maintain it :( I could
probably port all Jenkinsfiles into Jenkinsfile Runner so that reporting is
retained, but I am allowed to do so due to arrangements with my employer

stffrdhrn, 2021-06-27:
I think that will be.  We have what we need now, I will disable it for now as its flagging our commits as broken.  We can always re-enable when resources come back.