openrisc/mor1kx #127: spr_bus_ack_o reset issue

Harshitha172000, 2021-07-08:
This issue is applicable to the mor1kx_icache module. After resetting, the state moves to IDLE but spr_bus_ack_o can be high if spr strobe/write signals stays high during reset. This is possible as spr stb/we signals are not reset in the ctrl module.

Assertion Failed:
```
          always @(posedge clk)
               if ($past(rst) && f_past_valid)
                  assert (!spr_bus_ack_ic_i);
```
Trace showing issue:

[image]

stffrdhrn, 2021-07-09:
I posted a fix for this does it also fix formal?

Harshitha172000, 2021-07-09:
No, now I see this trace.  With the reset, the state switches from INVALIDATE to IDLE and keeps spr_bus_ack_o high.

[image]

stffrdhrn, 2021-07-09:
OK, it doesn't cater for the case when `spr_bus_ack_o` is high to start with.

stffrdhrn, 2021-07-10:
I pushed another update, how does it look now?

Harshitha172000, 2021-07-10:
It passes the test, the issue is resolved.

closed 2021-07-10 by Harshitha172000