openrisc/mor1kx #135: SPR property fails when debug is enabled

Harshitha172000, 2021-08-06:
SPR read and write requests are mutually exclusive. Both spr_read_access and spr_write_acess can’t be high at the same time. When the Debug unit is enabled, this property fails.

**Assertion** (Check applied to mor1kx_ctrl_cappuccino.v)
```
always @*
   assert ($onehot0 ({spr_read_access, spr_write_access}));
```
**Trace:**

[image]

stffrdhrn, 2021-08-15:
can you put a link to the code, or mention the module where this is happening?

Harshitha172000, 2021-08-16:
mor1kx_ctrl_cappuccino.v

Code:
https://github.com/openrisc/mor1kx/blob/master/rtl/verilog/mor1kx_ctrl_cappuccino.v#L1644#L1651