openrisc/mor1kx #136: spr_ack is high when there is no mtspr/mfspr instructions

Harshitha172000, 2021-08-06:
In the control module, signal `spr_access` is driven by the `spr_group` which is dependent on the address computed in the execution stage i.e., `ctrl_alu_result_i`. When we don’t have spr instructions but the last five bits of address `ctrl_alu_result_i` match to any of SPR groups then `spr_access` will get updated. Due to which `spr_access_valid` and `spr_ack` are driven when there are no mtspr/mfspr opcode signals.

Assertion:

```
always @(posedge clk)
   if ($onehot(spr_access) && !$past(rst) && f_past_valid)
      assert (ctrl_op_mtspr_i | ctrl_op_mfspr_i);
```

**Trace:**
[image]

stffrdhrn, 2021-08-15:
can you put a link to the code, or mention the module where this is happening?

Harshitha172000, 2021-08-16:
mor1kx_ctrl_cappuccino.v

Code:
https://github.com/openrisc/mor1kx/blob/master/rtl/verilog/mor1kx_ctrl_cappuccino.v#L1653#L1656