openrisc/mor1kx #139: The carry flag implementation is not correct for subtract instructions

Rahul-Kande, 2021-09-03:
The carry flag value is not as expected for some cases. For instance, with the following trigger input, since r1 > r3, carry flag should be reset, but it is actually being set. 

Issue location:  rtl/verilog/mor1kx_execute_alu.v

Triggering input: 
     //set r1=00020000 and r3=00002000
     l.sub    r4,r1,r3 
Expected output: 
               Carry flag = 0

mor1kx output: 
              Carry flag = 1

rkrajnc, 2021-09-06:
I can't test this right now, so I'm just guessing, but it seems the problem might be operator precedence in this line:
https://github.com/openrisc/mor1kx/blob/95eee0596a160ffdfd8ee6bc8b88268b2e49ec5e/rtl/verilog/mor1kx_execute_alu.v#L205

Besides, shouldn't the statement be using logical operators instead of bitwise?

skristiansson, 2021-09-06:
> I can't test this right now, so I'm just guessing, but it seems the
problem might be operator precedence in this line:
>
https://github.com/openrisc/mor1kx/blob/95eee0596a160ffdfd8ee6bc8b88268b2e49ec5e/rtl/verilog/mor1kx_execute_alu.v#L205

No, the precedence is correct, for subtraction you want to do the (2
complement) calculation:
a - b => a + ~b + 1
carry_in is used as the '1', thus it always need to be set for subtraction.

> Besides, shouldn't the statement be using logical operators instead of
bitwise?

They are all 1-bit signals, so bitwise and logical obtain the same result.