openrisc/mor1kx #140: The EPCR register is accessible from user mode

Rahul-Kande, 2021-09-03:
The OpenRISC specification requires that the EPCR register be accessible only from supervisor mode. 
 Bug location:  rtl/verilog/mor1kx_ctrl_cappuccino.v
 
 Triggering input: 
#include <stdio.h>
int main() {
  // enter user mode :
  asm volatile ( "l.ori r17,r0,0x0000  ");
  asm volatile ( "l.mtspr r0,r17,0x3806");

  asm volatile ( "l.mfspr r17,r0,0x11  ");
  asm volatile ( "l.andi r17,r17,-2    ");
  asm volatile ( "l.mtspr r0,r17,0x11  ");
  // padding the seed code
  asm volatile( "l.nop 0x0 ");
  asm volatile( "l.nop 0x0 ");
  asm volatile( "l.nop 0x0 ");
  asm volatile( "l.nop 0x0 ");
  asm volatile ("l.addi r1,r0,1");
  asm volatile ("l.mfspr r2,r0,32");
  asm volatile ("l.mtspr r0, r1, 32");
  return 0;
}

The or1ksim fails to execute the mfspr while the mor1kx implementation can successfully write into EPCR using the mtspr instruction. 

Please check this bug.

stffrdhrn, 2021-09-09:
Hello, thanks for your bug report and your mail.  
Mail:
https://lists.librecores.org/pipermail/openrisc/2021-September/003306.html

I think we will find a lot of registers that are missing protection in user mode.  It maybe good to list them all in one place.  Looking at the code I also see:
  - in mor1kx_ctrl_cappuccino, ESR, EVBAR, DMR1, DMR2, DSR, DRR
  - in mor1kx_pic PICMR, PICSR
  - ITLB registers
  - DTLB registers

stffrdhrn, 2021-09-09:
I think that wile fixing these we should add a test to our [or1k-tests](https://github.com/openrisc/or1k-tests) that verifies the correctness.