openrisc/mor1kx #141: Unable to write to the EEAR register from the supervisor mode.

Rahul-Kande, 2021-09-03:
The OpenRISC specification requires that the EEAR register be accessible from the supervisor mode but the mor1kx implementation does not have the option to write to EEAR with the mtspr instruction even from the supervisor mode. 
 
Issue location:  mor1kx_ctrl_cappuccino.v (https://github.com/openrisc/mor1kx/blob/master/rtl/verilog/mor1kx_ctrl_cappuccino.v), line 830 to 840.

Please check this bug.

stffrdhrn, 2021-09-09:
Hello, thanks for your bug report and your mail.   I see from your mail that you have been doing research can you explain how that works?  How did you find this?

Mail:
https://lists.librecores.org/pipermail/openrisc/2021-September/003306.html

 There are a few things where the mor1kx differs from the spec to make the implementation more efficient.   In this case we have no use case in existing code that requires EEAR to be written to so it was omitted.

Rahul-Kande, 2021-09-09:
Hi, 

We are developing hardware verification tools to detect bugs in RTL designs. In this case, we simulated mor1kx and [or1ksim ](https://github.com/openrisc/or1ksim) with the inputs generated from our tool and compared the trace outputs. We modified the tracing logic to output the values of all the GPRs and important SPRs. So, this is how we detected this bug.

stffrdhrn, 2021-09-11:
Wow, that is great, that sounds like "golden reference" ([example](http://www.syncad.com/web_manual_testbencher/test_bench_generator_main_index.html?golden_reference_models.htm)) verification.  This is/was something that is on my todo list to implement for mor1kx using or1ksim.  Do you think you will be able to contribute this back to openrisc?

Rahul-Kande, 2021-09-11:
Hi, 
Sure. Our work is yet to be published, so I cannot share the tool and its specific details right now. But, once we get our paper out, I can check with my advisor and get back to you.