openrisc/mor1kx #145: Implement write access to FPCSR if SR[SUMRA] is raised.

bandvig, 2022-01-13:
@stffrdhrn 
Stafford,
In framework of `https://openrisc.io/proposals/p17-user-mode-fpcsr` proposal, I've implemented write access to FPSCR if SR[SUMRA] is raised. Please find the modifications in `fpcsr_sumra` branch. The modifications are quite similar those I did for MAROCCHINO (in the similar branch). 
I hope you can use the branch for implementation FP support in glibc and linux.

stffrdhrn, 2022-01-14:
Hi @bandvig 

The change looks good. But I am thinking maybe we should not use Sr[sumra] and just make fpscr always accessible.  It will make the change easier and it seems like what other architecturesc do.

What do you think?  I can update the proposal

bandvig, 2022-01-15:
@stffrdhrn 
I have no objection. I also created `fpcsr_free_access` branches for both `mor1kx` and `or1k_marochino`. Modifications are trivial.

stffrdhrn, 2022-02-28:
Merged the fix

closed 2022-02-28 by stffrdhrn