openrisc/mor1kx #156: How to run mor1kx with verilator?

zhangkanqi, 2023-11-07:
Hi!
I want to run mor1kx with verilator, but I dodn't get the results I want directly from the document and readme. 

Specifically, in order to run with verilator, I need only **one** verilog file rather than many verilog files. But I don't know how to complie whole mor1kx  project to one verilog file. Could u give me some hints, please?

Looking forward to your reply:)

juliusbaxter, 2023-12-11:
I wonder if one of the FuseSoC builds of mor1kx has a Verilator target?

@stffrdhrn or @olofk do you know? Thanks!

stffrdhrn, 2023-12-12:
Hello, yes we can run mor1kx with verilator using the fusesoc build.  For example https://github.com/stffrdhrn/mor1kx-generic/blob/master/mor1kx-generic.core

This supports both verilator and Icarus.  You can run it with fusesoc, there are some examples in the mor1kx CI if you want to dig.

juliusbaxter, 2023-12-13:
Nice one, thanks Stafford, indeed that does contain support for building mor1kx with Verilator in an example SoC.

closed 2023-12-13 by juliusbaxter