openrisc/mor1kx #160: Synthesis Bugs in the LSU

rsar97, 2025-01-05:
Hi,

We found a output mismatch between the RTL and the netlist of the or1200 cpu output port (du_lsu_load_dat) orginating from the LSU module of the port. 
[image]
[image]

stffrdhrn, 2025-01-12:
Hi @rsar97 thanks for the bug report, do you have a solution for this or working on a patch?

What synthesis tool are you using?

rsar97, 2025-01-27:
Hi @stffrdhrn thank you for the response. We use Synopsys DC and Cadence Genus. We found these bugs in the netlist generated by both of the tools. We don't have a solution.